4IceG/luci-app-3ginfo-lite #106: error parsing fibocom fm350 data

SuperKali, 2024-11-22:
Hi,

I noticed that very often the fibocom returns as CSQ value 99, 99, this problem implies the entire dashboard of 3ginfo not showing the data:

```json
"csq":"-",
"signal":"0",
```

```bash
+ awk '-F[, ]' '/^\+CSQ/ {print $2}'
+ CSQ=99
+ '[' x99 '=' x ]
+ '[' 99 -ge 0 -a 99 -le 31 ]
+ CSQ=-
+ CSQ_PER=0
+ '[' -z - ]
+ '[' -z 0 ]

```
[image]

Is there a way to solve it?

Thank you

SuperKali, 2024-11-23:
> Find another command to get CSQ

The problem that there is no other way to get the data, the only way is via AT commands, which is causing this problem, I had to put the value 1 on both variables to keep it from looping, this however will report 1% signal when the value is invalid, but it doesn't trigger the 3ginfo interface loop. I don't understand if it's a module problem or a firmware problem, I updated it but nothing changed

greg33331, 2025-01-02:
Same, 
with Fibocom FM350-GL module sporadically the connection is interrupted and it is not possible to view the metrics.
Instead through AT commands it is possible to request the information continuously.
It is possible to fix the problem?
Thanks!

sjjcyn, 2025-01-04:
原因是fm350的at命令无法获取csq的值，只要修改一下/usr/share/3ginfo-lite/3ginfo.sh这个脚本就可以使用。
[3ginfo.sh.txt](https://github.com/user-attachments/files/18305271/3ginfo.sh.txt)

sjjcyn, 2025-01-04:
> 原因是fm350的at命令无法获取csq的值，只要修改一下/usr/share/3ginfo-lite/3ginfo.sh这个脚本就可以使用。 [3ginfo.sh.txt](https://github.com/user-attachments/files/18305271/3ginfo.sh.txt)

openwrt 需要安装bc软件包 opkg install bc

kowi4, 2025-02-28:
I also got this one with my modem. I quickly fixed that by putting 1 if CSQ is out of range. I think the proper fix would be to allow undefined value and just let the user see all the rest of parameters even if signal is empty.