4IceG/luci-app-3ginfo-lite #107: Feature Request: Separate 5G signal readings on 5G-NSA mode

xhikarishii, 2024-12-04:
Hi @4IceG 

I hope you are having a good week, I don't know if this is asking too much, but having this feature on your package would be nice. 

I am willing to contribute to this feature to push through.

f1vefour, 2024-12-08:
You mean displaying the 4G and 5G signals independently with two signal bars? If so I definitely like this idea.

xhikarishii, 2024-12-08:
@f1vefour yes exactly

4IceG, 2024-12-08:
Hi @xhikarishii @f1vefour ,
When 3ginfo was created, hardly anyone had heard of LTE-A/5G. Looking at how many packages have been added for LuCI over the years (for mobile internet), it's sad that everything is based on the work of me and Cezary from the eko.one.pl forum (I don't count the ROOTer team because it's a dedicated project).

The package needs to be rewritten to:
- add support for multiple modems from tabs
- add support for more signal data/parameters
- maintain compatibility with modems from LTE to 5G

I don't think anyone will do it except me.. again..