4IceG/luci-app-3ginfo-lite #108: EP06-E wrong sinr value

1alessandro1, 2024-12-08:
[image]

Hello, as you can see from the picture the SINR value is reported badly, on the right side you have the AT command to check SINR and on the left side the javascript reports a negative value way below the limits. The firmware on the EP06 I'm using is EP06ELAR04A22M4G

This is the example of the string that reports the channel info:

`+QENG: "servingcell","NOCONN","LTE","FDD",222,88,4CAA71F,28,1500,3,3,3,77B8,-84,-9,-56,21,35`

where:

222 = MNC
88 = MCC
4CAA71F = eNB (hex)
28 = Cell ID
1500 = EARFCN

-84 = RSRP 
-9 = RSRQ
-56 = RSSI
21 = SINR

1alessandro1, 2024-12-08:
Seems like this awk call is the issue. Looking at the script used for EP06E 2c7c0306

[image]

1alessandro1, 2024-12-08:
The other problem is present when asking the modem "QCAINFO".  The second band, the one who is aggregated, is always reported with SINR = 0 

[image]

The script will parse that result, multiply by 0.2 and subtract 20, displaying a negative SINR (wrong value). We should take PCC SINR for reference.

4IceG, 2024-12-08:
Hi,
@1alessandro1 I don't believe in what modem shows (in at commands) because the indications may differ significantly depending on the modem's fw.

I understand that you checked documentation and asked about this value on Quectel forum?

1alessandro1, 2024-12-08:
I've checked the documentation: 
[Quectel_EP06&EG06&EM06_AT_Commands_Manual_V1.0.pdf](https://github.com/user-attachments/files/18054082/Quectel_EP06.EG06.EM06_AT_Commands_Manual_V1.0.pdf)

page 86:

```
AT+QENG="servingcell"
Response
In LTE mode:
+QENG: "servingcell",<state>,"LTE",<is_tdd>,<mcc>,<m
nc>,<cellid>,<pcid>,<earfcn>,<freq_band_ind>,<ul_band
width>,<dl_bandwidth>,<tac>,<rsrp>,<rsrq>,<rssi>,<sin
r>,<srxlev>
```

The penultimate value is SINR for the output of `AT+QENG="servingcell"`.

The same does not apply to `AT+QCAINFO`, where this time the last value is SINR

```
+QCAINFO
Response:

PCC”,<freq>,<bandwidth>,<band>,<pcell_state>,<pcid>,
<rsrp>,<rsrq>,<rssi>,<sinr>
“SCC”,<freq>,<bandwidth>,<band>,<scell_state>,<pcid>,
<rsrp>,<rsrq>,<rssi>,<sinr>
```

Additionally, I know what SINR values to expect for each band. I've connected this EP06 to my personal external antenna, and I know that on B3 (1800MHz) I expect around 15-17 dB for SINR. I've connected also another modem to check this, and there's nothing wrong with the setup (a -17SNR would suggest that something is broken, while it really isn't)

4IceG, 2024-12-08:
page 88:

[image]

1alessandro1, 2024-12-09:
I might have found the issue. The conversion formula is

 sinr = (1/5)*(value from AT)*10 - 20

This is what is reported in the documentation: 

[image]

Page 107:
[Quectel_RM5x0N_RM521F_Series_AT_Commands_2024_02_07.pdf](https://github.com/user-attachments/files/18054519/Quectel_RM5x0N_RM521F_Series_AT_Commands_2024_02_07.pdf)

I've compared the values from the DIAG port with the values reported from the AT serial port

```
+QENG: "servingcell","NOCONN","LTE","FDD",222,88,AB6A40C,324,3350,7,5,5,7E11,-10              7,-11,-75,11,11,160,-
```

This is what airscreen shows: 
[image]

The formula would be: Real SINR = 13*0.2 * 10 - 20 = 6 dB

1alessandro1, 2024-12-10:
Hello, any update on this?

4IceG, 2024-12-11:
I'm already lost, I don't know for which modem you are looking for an error.

1alessandro1, 2024-12-23:
Hello, the quectel documentation for the EP06-E reports that the second to least value from AT+QENG="servingcell".

I cross checked what values are reported from the RM521F-GL modem in LTE mode: here it's the fourth to least value.

[image]

My hypotesis is that the values reported follow the same pattern, and can be decoded with the same formula, written only in RM521F-GL manual afaik. For additional redundacy, these values are also compared in the same location with the data parsed from the DIAG port, using AirScreen software.

Let's choose a preferred band: B38. This band will be the one that we have to find SINR. 
This is what the EP06E firmware (EP06ELAR04A22M4G) is reporting: 

[image]

The same exact setup was repeated by swapping only the modem. Here is what the RM521F-GL is saying: 

[image]

And looking at the code, your script is configured correctly for 2c7c0801: 

[image]

Let's check what Qualcomm DIAG port is reporting (does not use AT commands): 
The value we are expecting is 23*0.2*10 - 20 = 26 dB

[image]

Indeed your script is okay for rm521f, on the same bts, same cell, same frequency:

[image]

Conclusion: the ep06e is reporting 20 (that should be interpreted as 20*0.2*10 - 20 )

1alessandro1, 2024-12-29:
@4IceG any update on this? Did I clarify why the EP06 A22 firmware uses the same syntax as the newer 5G modems when replying to AT+QENG="servingcell" ?

1alessandro1, 2025-01-05:
I will attach the fixed 2c7c0306 file for EP06ELAR04A22M4G for anybody who has the 2023 firmware on this old modem from 2018 (maybe back then, they used 0-250 range, now even in front of the tower there is no way to get that far (to 250) even when I know DIAG port logging reports that value. Hope it helps somebody if it won't get merged in master.
[2c7c0306.zip](https://github.com/user-attachments/files/18308846/2c7c0306.zip)

1alessandro1, 2025-02-27:
@4IceG what happened? Are there still doubts after all the additional information I provided you about the SINR values that I expect to see (and I double checked with DIAG logs) and what is actually reported by your script?

giorez, 2025-04-24:
thanks @1alessandro1 I solved the problem on my EP06-E