4IceG/luci-app-3ginfo-lite #124: 3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/12d1155e: line 100: arithmetic syntax error

ZAV-8, 2025-05-22:
Hello,
script 3ginfo.sh and 3ginfo-lite menu works only in LTE mode (Network mode in interface settings set in Prefer LTE or LTE).
Any another mode (Prefer UMTS or UMTS/GPRS and etc) in interface settings give error script

/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/12d1155e: line 100: arithmetic syntax error

and
OpenWrt 3ginfo-lite page give next errors:
SyntaxError Unexpected end of JSON input
Error: Unexpected end of JSON input

luci-app-3ginfo-lite version is 1.0.75-r20250517

ZAV-8, 2025-05-23:
Debug output:

+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB2
+ '[' -z /dev/ttyUSB2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?

+CPIN: READY

+CSQ: 13,0

+COPS: 0,0,"MTS BY",2

+COPS: 0,2,"25702",2

+CREG: 2,1,"00D3","00D09A81",6

OK
'
+ CONFIG=modemdefine
+ + + grep -o '@modemdefine\[[0-9]*\]\.modem'
uci show modemdefine
wc+  -l
xargs
+ MODEMZ=0
+ '[[' 0 -gt 1 ]]
+ '[[' 0 -eq 0 ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=
+ '[[' 0 -eq 1 ]]
+ '[' -z  ]
+ getpath /dev/ttyUSB2
+ basename /dev/ttyUSB2
+ devname=ttyUSB2
+ readlink -f /sys/class/tty/ttyUSB2/device
+ devpath=/sys/devices/platform/ahb/1b000000.usb/usb1/1-1/1-1:1.2/ttyUSB2
+ P=/sys/devices/platform/ahb/1b000000.usb/usb1/1-1
+ PORIG=/sys/devices/platform/ahb/1b000000.usb/usb1/1-1
+ getpath /dev/console
+ basename /dev/console
+ devname=console
+ readlink -f /sys/class/usbmisc/console/device/
+ devpath=
+ P=
+ '[' x/sys/devices/platform/ahb/1b000000.usb/usb1/1-1 '==' x ]
+ getpath /dev/ptmx
+ basename /dev/ptmx
+ devname=ptmx
+ readlink -f /sys/class/usbmisc/ptmx/device/
+ devpath=
+ P=
+ '[' x/sys/devices/platform/ahb/1b000000.usb/usb1/1-1 '==' x ]
+ getpath /dev/tty
+ basename /dev/tty
+ devname=tty
+ readlink -f /sys/class/tty/tty/device
+ devpath=/sys/devices/virtual/tty/tty/device
+ P=/sys/devices/virtual/tty
+ '[' x/sys/devices/platform/ahb/1b000000.usb/usb1/1-1 '==' x/sys/devices/virtual/tty ]
+ getpath /dev/ttyS0
+ basename /dev/ttyS0
+ devname=ttyS0
+ readlink -f /sys/class/tty/ttyS0/device
+ devpath=/sys/devices/platform/ahb/ahb:apb/18020000.uart
+ P=/sys/devices/platform/ahb
+ '[' x/sys/devices/platform/ahb/1b000000.usb/usb1/1-1 '==' x/sys/devices/platform/ahb ]
[... 531 lines cut ...]
+ XFR='at^hfreqinfo?

'
+ + wc -c
echo -n 'at^hfreqinfo?

'
+ XFRCOUNT=16
+ '[' -n 'at^hfreqinfo?

' ]
+ '[' 16 -gt 15 ]
+ + + echo 'at^hfreqinfo?

'
awk -F '[:]'xargs '/^\^HFREQINFO:/{print $2}'

+ TBX=
+ + cutecho -d  ,
 -f3-3
+ TB=
+ + cutecho -d  ,
 -f6-6
+ TDL=
/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/12d1155e: line 94: arithmetic syntax error

4IceG, 2025-05-23:
Replace this file /usr/share/3ginfo-lite/modem/usb/12d1155e and check if it will be OK.
[12d1155e.zip](https://github.com/user-attachments/files/20417177/12d1155e.zip)

ZAV-8, 2025-05-24:
My report after replace 12d1155e.
Modem in LTE mode
/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/12d1155e: line 115: syntax error: unexpected "fi"
I deleted string 115 "fi" from fle and script started.

Modem in UMTS mode
/usr/share/3ginfo-lite/modem/usb/12d1155e: line 96: arithmetic syntax error

Debug info:

```
+ RES=/usr/share/3ginfo-lite
+ /usr/share/3ginfo-lite/detect.sh
+ DEVICE=/dev/ttyUSB2
+ '[' -z /dev/ttyUSB2 ]
+ O=
+ '[' -e /usr/bin/sms_tool ]
+ sms_tool -D -d /dev/ttyUSB2 at 'AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?'
+ O='AT+CPIN?;+CSQ;+COPS=3,0;+COPS?;+COPS=3,2;+COPS?;+CREG=2;+CREG?

+CPIN: READY

+CSQ: 13,0

+COPS: 0,0,"MTS BY",2

+COPS: 0,2,"25702",2

+CREG: 2,1,"00D3","00D09A81",6

OK
'
+ CONFIG=modemdefine
+ + + grepuci -o '@modemdefine\[[0-9]*\]\.modem' show
 modemdefine
+ wc -l
xargs
+ MODEMZ=0
+ '[[' 0 -gt 1 ]]
+ '[[' 0 -eq 0 ]]
+ uci -q get '3ginfo.@3ginfo[0].network'
+ SEC=modemwan
+ '[[' 0 -eq 1 ]]
+ '[' -z modemwan ]
+ CONN_TIME=-
+ RX=-
+ TX=-
+ + grep '"up": true'
ifstatus modemwan
+ NETUP='	"up": true,'
+ '[' -n '	"up": true,' ]
+ uci -q -P /var/state/ get network.modemwan.connect_time
+ CT=
+ '[' -z ]
+ + + ifstatusxargs modemwan

awk '-F[:,]' '/uptime/ {print $2}'
+ CT=217
+ '[' '!' -z 217 ]
+ expr 217 / 60 / 60 / 24
+ D=0
+ expr 217 / 60 / 60 '%' 24
+ H=0
+ expr 217 / 60 '%' 60
+ M=3
+ expr 217 '%' 60
+ S=37
+ printf '%dd, %02d:%02d:%02d' 0 0 3 37
+ CONN_TIME='0d, 00:03:37'
+ date '+%s'
[... 365 lines cut ...]
'
+ + wc -c
echo -n 'at^hfreqinfo?

'
+ XFRCOUNT=16
+ '[' -n 'at^hfreqinfo?

' ]
+ '[' -n 16 ]
+ '[' 16 -gt 15 ]
+ + + xargsawk
 -Fecho 'at^hfreqinfo?

' '[:]'
 '/^\^HFREQINFO:/{print $2}'
+ TBX=
+ + cut -d , -f3-3
echo 
+ TB=
+ + cut -decho ,  -f6-6

+ TDL=
/usr/share/3ginfo-lite/3ginfo.sh: /usr/share/3ginfo-lite/modem/usb/12d1155e: line 96: arithmetic syntax error
```

ZAV-8, 2025-05-24:
I think script can't get value DL, UL (and also BAND) in UMTS mode.
I replaced it on integer and script started (these and signal quality values are absent)

[image]

result

[image]

thank you for attention

4IceG, 2025-05-24:
Script needs fixing, because for WCDMA the modem does not return much of the data required by LuCI. I will look at it later.

4IceG, 2025-05-29:
It is OK, it shows only what is available.