4IceG/luci-app-3ginfo-lite #126: Consider supporting uqmi device interface

cybermaus, 2025-06-14:
My device, a tp-link mr200, has an internal USB modem that is internally a Android device, and I understand this is not uncommon anymore nowadays. it has the ```/dev/cdc-wdm0``` device, but I am unable to make it present additional ```/dev/ttyUSB``` devices.

However, all needed data is presented in JSON format through the ```uqmi``` utility, and I wonder since that utility is broader in use for all QMI devices if that is not also a good generic status exchange protocol.

some examples below, but first the raw USB data:
```
root@OpenWrt:~# lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux 6.6.86 ehci_hcd EHCI Host Controller
Bus 001 Device 002: ID 05c6:9025 Android Android
```

```
root@OpenWrt:~# cat /sys/kernel/debug/usb/devices

T:  Bus=01 Lev=00 Prnt=00 Port=00 Cnt=00 Dev#=  1 Spd=480  MxCh= 1
B:  Alloc=  0/800 us ( 0%), #Int=  0, #Iso=  0
D:  Ver= 2.00 Cls=09(hub  ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=1d6b ProdID=0002 Rev= 6.06
S:  Manufacturer=Linux 6.6.86 ehci_hcd
S:  Product=EHCI Host Controller
S:  SerialNumber=101c0000.ehci
C:* #Ifs= 1 Cfg#= 1 Atr=e0 MxPwr=  0mA
I:* If#= 0 Alt= 0 #EPs= 1 Cls=09(hub  ) Sub=00 Prot=00 Driver=hub
E:  Ad=81(I) Atr=03(Int.) MxPS=   4 Ivl=256ms

T:  Bus=01 Lev=01 Prnt=01 Port=00 Cnt=01 Dev#=  2 Spd=480  MxCh= 0
D:  Ver= 2.00 Cls=00(>ifc ) Sub=00 Prot=00 MxPS=64 #Cfgs=  1
P:  Vendor=05c6 ProdID=9025 Rev= 3.18
S:  Manufacturer=Android
S:  Product=Android
S:  SerialNumber=9bd61336
C:* #Ifs= 6 Cfg#= 1 Atr=a0 MxPwr=500mA
I:* If#= 0 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=ff Prot=ff Driver=(none)
E:  Ad=81(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=01(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 1 Alt= 0 #EPs= 2 Cls=ff(vend.) Sub=42 Prot=01 Driver=(none)
E:  Ad=02(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=82(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 2 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=84(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=83(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=03(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 3 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=00 Prot=00 Driver=(none)
E:  Ad=86(I) Atr=03(Int.) MxPS=  10 Ivl=32ms
E:  Ad=85(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=04(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 4 Alt= 0 #EPs= 3 Cls=ff(vend.) Sub=ff Prot=ff Driver=qmi_wwan
E:  Ad=88(I) Atr=03(Int.) MxPS=   8 Ivl=32ms
E:  Ad=87(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=05(O) Atr=02(Bulk) MxPS= 512 Ivl=0ms
I:* If#= 5 Alt= 0 #EPs= 2 Cls=08(stor.) Sub=06 Prot=50 Driver=(none)
E:  Ad=89(I) Atr=02(Bulk) MxPS= 512 Ivl=0ms
E:  Ad=06(O) Atr=02(Bulk) MxPS= 512 Ivl=125us
root@Camper_Maurits:~#
```

Some ```uqmi``` examples:
```
root@OpenWrt:~# uqmi -d /dev/cdc-wdm0 --get-profile-settings 3gpp,1
{
        "apn": "internet",
        "pdp-type": "ipv4",
        "username": "",
        "password": "",
        "auth": "none",
        "no-roaming": false,
        "apn-disabled": false
}
root@OpenWrt:~# uqmi -d /dev/cdc-wdm0 --get-current-settings
{
        "pdp-type": "ipv4",
        "ip-family": "ipv4",
        "mtu": 1500,
        "ipv4": {
                "ip": "10.195.7.33",
                "dns1": "212.2.127.254",
                "dns2": "212.2.96.52",
                "gateway": "10.195.7.34",
                "subnet": "255.255.255.252"
        },
        "ipv6": {

        },
        "domain-names": {

        }
}
root@OpenWrt:~# uqmi -d /dev/cdc-wdm0 --set-network-modes lte
root@OpenWrt:~# uqmi -d /dev/cdc-wdm0 --get-signal-info
{
        "type": "lte",
        "rssi": -73,
        "rsrq": -15,
        "rsrp": -107,
        "snr": 5.400000
}
root@OpenWrt:~# uqmi -d /dev/cdc-wdm0  --get-serving-system
{
        "registration": "registered",
        "radio_interface": [
                "lte"
        ],
        "plmn_mcc": 204,
        "plmn_mnc": 16,
        "plmn_description": "Or▒▒\u0006",
        "roaming": true
}
```

MADRAFi, 2025-06-17:
I use modem RM520N-GL. It creates device /dev/wwan0mbim0 that can be interact with. Other devices do not work.
Can you add support for uqmi for it?

# uqmi -m -d /dev/wwan0mbim0 --get-system-info
{
        "wcdma": {
                "service_status": "none",
                "true_service_status": "none",
                "preferred_data_path": false
        },
        "lte": {
                "service_status": "available",
                "true_service_status": "available",
                "preferred_data_path": false,
                "domain": "cs-ps",
                "service": "cs-ps",
                "roaming_status": "off",
                "forbidden": false,
                "mcc": "260",
                "mnc": "03",
                "tracking_area_code": 58145,
                "enodeb_id": 181606,
                "cell_id": 15,
                "voice_support": true,
                "ims_voice_support": true,
                "cell_access_status": "all calls",
                "registration_restriction": 0,
                "registration_domain": 0,
                "5g_nsa_available": true,
                "dcnr_restriction": false
        },
        "5gnr": {
                "service_status": "none",
                "true_service_status": "none",
                "preferred_data_path": false
        }
}

# uqmi -m -d /dev/wwan0mbim0 --get-lte-cphy-ca-info
{
        "primary": {
                "cell_id": 454,
                "channel": 6200,
                "band": 20,
                "frequency": 800,
                "duplex": "FDD",
                "bandwidth": "10"
        }
}

# uqmi -m -d /dev/wwan0mbim0 --get-serving-system
{
        "registration": "registered",
        "radio_interface": [
                "lte"
        ],
        "plmn_mcc": 260,
        "plmn_mnc": 3,
        "plmn_description": "Orange",
        "roaming": false
}

# uqmi -m -d /dev/wwan0mbim0 --get-signal-info
{
        "type": "lte",
        "rssi": -65,
        "rsrq": -12,
        "rsrp": -94,
        "snr": 6.000000
}

4IceG, 2025-06-21:
Hi,

Currently I am not planning to add anything to the package.
I plan to rewrite the package to show more data, but what will come out of that I don't know.