4IceG/luci-app-3ginfo-lite #128: For LTE only sim cards it show as registration denid

nooblk-98, 2025-06-29:
For LTE only sim cards it show as registration denid can you fix this ?

internet working fine but cell informations not showing also 

my modem RG500QEAAAR13A01M4G

4IceG, 2025-06-30:
The package only shows what the scripts return, I don't see this data, so I don't know where the problem is.

MalinrRuwan, 2025-08-03:
@nooblk-98 Can you tell me how you fixed the issue?

4IceG, 2025-08-04:
> got aw1000 send me the ipk file. I saw your repo. Nice work brother. [@nooblk-98](https://github.com/nooblk-98) send it to my email [maliniqrub@gmail.com](mailto:maliniqrub@gmail.com)

You've seen the repository, installing packages couldn't be simpler. Add a repository https://github.com/4IceG/Modem-extras or download the package manually.

closed 2026-09-05 by nooblk-98