4IceG/luci-app-3ginfo-lite #131: Add Quectel MHI driver parsing

stich86, 2025-08-10:
Hi @4IceG,

i'm playing with a CPE that supporto only PCIE module, so I got an RM520N-GLAP (no the AA version). Using Quectel MHI driver, is not possible to see all informations.

I've edited `3ginfo.sh` and `getdevicevendorproduct()` function adding this entry into the case statement:

```
        'mhi_DUN'*)
            devpath=$(find /sys/devices -name "$devname" -type d 2>/dev/null | head -1)
            target_dir=${devpath%/*/*/*}
            V=$(cat "$target_dir/vendor")
            D=$(cat "$target_dir/device")
            echo "pci/${V/0x/}${D/0x/}"
	    ;;
```

`/dev/mhi_DUN` is the AT port exposed by Quectel MHI driver. The i've create under `modem/pci` folder the usual file with VIDPID, for Quectel RM520N (using the usb file), changing only the last part to display currently working protocol:

```
# Protocol
# DRIVER=QMI_WWAN & DRIVER=CDC_MBIM
if [ -d "/sys/class/mhi_uci_q" ]; then
    if ls /sys/class/mhi_uci_q/mhi_QMI0 > /dev/null 2>&1; then
        PROTO="qmi"
    elif ls /sys/class/mhi_uci_q/mhi_MBIM* > /dev/null 2>&1; then
        PROTO="mbim"
    else
        PROTO="unknown"
    fi
fi
```

Let me know if you want a PR 

Thanks!

4IceG, 2025-08-11:
Hi @stich86 ,
You can make a PR if you want 😏, but I don't know when the new package/repository update will be. Right now, I'm focusing on preparing the new package.

4IceG, 2025-11-21:
Hi, @stich86 ,
I repeat the question :)

SuperKali, 2025-11-21:
@stich86 @4IceG here https://github.com/SuperKali/wwan-packages/blob/master/utils/quectel-cm/files/quectel.sh

I have to push local changes, will do soon, currently was pushed the old changes that should be work, i reworked autoreconnect but untested on pcie module